adafruit/Adafruit-PN532 #9: Only receiving 31 bytes from ISO14443A card

juliabory, 2015-07-06:
I have activated debug output and I see that after 31 byte of data always 0xFF coming from i2c_recv() in line 1752. Has anyone tried to receive more than 31 byte from a java smart card? I am happy about any hint.
Output Example:
Reading:  0x0 0x0 0xFF 0x46 0xBA 0xD5 0x41 0x0 0x4 0xAF 0x7F 0x4F 0x60 0x31 0x6D 0xB9 0x76 0x8C 0x88 0x9D 0x7B 0x32 0xCC 0x25 0x63 0x2D 0xE9 0xF6 0x75 0x67 0x63 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF

microbuilder, 2015-07-06:
I2C is limited to 32 bytes, so you'll probably need to switch to SPI for larger transfers.

closed 2015-07-06 by microbuilder

juliabory, 2015-07-07:
Thanks for the reply. We increased the i2c buffer to 128 byte and its working for us now. Switching to SPI wouldn't have been so easy for us und would eventually have the same problem. So we have done it that way.

jl-yang, 2015-10-26:
@juliabory  Hi I also found the i2c buffer too small for a NDEF message, could you tell me how to increase it to 128 byte? Because when I increased BUFFER_LENGTH to a bit larger than 32, the PN532 just failed.