adafruit/Adafruit-PN532 #10: Change default protocol to I2C in examples

tobiasebsen, 2015-07-24:
It's a bit strange that the default behaviour of the board is to use I2C (without soldering pads SEL0 and SEL1), but all the examples uses SPI (if not modifying the sketch). This means that users will not get the board working unless doing one of the two modifications.

pfeerick, 2015-07-26:
@tobiasebsen, I think that is due to the [breakout being in SPI mode by default](https://learn.adafruit.com/adafruit-pn532-rfid-nfc/breakout-wiring), and the [shield in I2C by default](https://learn.adafruit.com/adafruit-pn532-rfid-nfc/shield-wiring)? Anyway, I came here for to log a different issue, which you might be able to help with... did you also have to to change the following lines from 2,3 to 4,5? As I was testing a elechouse pn532 breakout with this code... and whilst it still doesn't work with the adafruit library, it didn't even detect the chip until I made that seemingly important change!

```
#define PN532_IRQ   (2)
#define PN532_RESET (3)  // Not connected by default on the NFC Shield
```

caternuson, 2023-02-13:
Example code modification is just something that might be needed. Esp. for boards with multiple interfaces and various other optional pins, like IRQ, reset, etc. Writing specific example for all possible permutations is not practical.

closed 2023-02-13 by caternuson