adafruit/Adafruit-PN532 #12: Warnings on compile

SethArchambault, 2015-09-24:
When Running readMifare I get these warnings:

/Users/watchmyfeet/Documents/Arduino/libraries/Adafruit-PN532-master/Adafruit_PN532.cpp: In member function 'uint8_t Adafruit_PN532::mifareclassic_WriteNDEFURI(uint8_t, uint8_t, const char*)':
/Users/watchmyfeet/Documents/Arduino/libraries/Adafruit-PN532-master/Adafruit_PN532.cpp:1048:53: warning: narrowing conversion of '(((int)len) + 5)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
   uint8_t sectorbuffer1[16] = {0x00, 0x00, 0x03, len+5, 0xD1, 0x01, len+1, 0x55, uriIdentifier, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
                                                     ^
/Users/watchmyfeet/Documents/Arduino/libraries/Adafruit-PN532-master/Adafruit_PN532.cpp:1048:72: warning: narrowing conversion of '(((int)len) + 1)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
   uint8_t sectorbuffer1[16] = {0x00, 0x00, 0x03, len+5, 0xD1, 0x01, len+1, 0x55, uriIdentifier, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
                                                                        ^
/Users/watchmyfeet/Documents/Arduino/libraries/Adafruit-PN532-master/Adafruit_PN532.cpp: In member function 'uint8_t Adafruit_PN532::ntag2xx_WriteNDEFURI(uint8_t, char*, uint8_t)':
/Users/watchmyfeet/Documents/Arduino/libraries/Adafruit-PN532-master/Adafruit_PN532.cpp:1416:8: warning: narrowing conversion of '(((int)len) + 5)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
     len+5,        /* Payload Length (not including 0xFE trailer) */
        ^
/Users/watchmyfeet/Documents/Arduino/libraries/Adafruit-PN532-master/Adafruit_PN532.cpp:1419:8: warning: narrowing conversion of '(((int)len) + 1)' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
     len+1,        /* Payload len */
        ^

and also these:

Sketch uses 8,960 bytes (27%) of program storage space. Maximum is 32,256 bytes.
Global variables use 1,067 bytes (52%) of dynamic memory, leaving 981 bytes for local variables. Maximum is 2,048 bytes.
avrdude: can't open input file /var/folders/08/1q3506q14bv8ycjvjc9zzrk00000gn/T/build279b921e1715ac438203aef4b85f66b2.tmp/readMifare.ino.hex: No such file or directory
avrdude: read from file '/var/folders/08/1q3506q14bv8ycjvjc9zzrk00000gn/T/build279b921e1715ac438203aef4b85f66b2.tmp/readMifare.ino.hex' failed

SethArchambault, 2015-09-24:
Adding this line to the top of the examples seems to help:
#include <stdint.h>
#include <stdbool.h>

But not sure.. after I removed them it continued to work..

glop102, 2016-02-26:
Don't worry about the errors. The compiler is paranoid.

ladyada, 2020-02-24:
fixed in 1.2.0+

closed 2020-02-24 by ladyada