adafruit/Adafruit-PN532 #13: 100% uC use

Thebobcachelot, 2015-10-06:
This library use 100% uc task While waiting for a nfc tag, the timeout in the library don't let possibilities to multitask with this librairy easily. As I know , a librarie should Never going into a almost infinit loop inside of itself.

glop102, 2016-02-26:
I have run into this problem myself. I am adding in support for the Felica style cards at the moment and the library waits for each card type to be read in it's own turn. This makes it impossible to support both cards at the same time.

I am thus forced to add a timeout function to the library when asking it to find a card. Lets hope i don't break everything...

Thebobcachelot, 2016-02-26:
There is a place in the library where yhe timeout is too long, i changed
the value and it work fine after

glop102, 2016-02-26:
thanks for the comment. I never noticed the timeout option. Thanks, it has solved all my problems.

caternuson, 2023-02-13:
Closing. Sounds resolved.