adafruit/Adafruit-PN532 #17: No delay needed when reading bytes from Wire library.

Koepel, 2016-01-05:
In line 1604 is a delay(1) ;
That is after the WIRE.requestFrom() and after the first i2c_recv(), and before reading each byte of the following data.
However, that delay is not needed, since the Wire.requestFrom() is a complete I2C transmission on its own. All the data has been read and it waiting in a buffer in the Wire library.

mevst, 2016-01-20:
I've tested this change trying to improve performance of NTAG memory reading.
The removal of the delay breaks the tag read functionality. This happens because the library does not implement handshake with PN512 chip consistently. 
Before trying to read from PN512 the host should check whether PN512 is ready. This is implemented correctly in Adafruit_PN532::sendCommandChackAck() but not in Adafruit_PN532::readdata() where delay() calls are placed here and there instead.

Koepel, 2016-01-20:
The isready() when using I2C, checks for the IRQ line. That line should be connected or course.

The call to Wire.requestFrom() is a complete I2C transmission. After it returns, the Wire.read() reads bytes from the buffer in the Wire library. The delay could only have some influence if the next communication with the chip is delayed.

Are you sure we are talking about the same delay ?
It is the delay(1) after WIRE.requestFrom() and during reading i2c_recv(), exept for the first i2c_recv().

mevst, 2016-01-20:
I am talking about the same delay. When I removed it I started to get wrong data in ntag2xx_ReadPage(). 

Debugging this I've found none consistent handshake implementation.
Please take a look into PN532 user manual (can be found from Adafruit website), page 42.
Host should not try to read data from PN532 until IRQ line gets low. 
As you can see from the library code there is no IRQ check in readdata(). This means that Wire.requestFrom () can be called too early. PN532 just has no valid data at that moment and responds with not ready bit in first byte (this byte is ignored in readdata() and dummy data.

I agree with you that any delays after the return from Wire.requestFrom() can not effect to the data already buffered by the function. 

Probably the delay(1) cause timing change in sendCommandCheckAck() which uses readdata() to read ACK frame from PN532. The second call of readdata within for example ntag2xx_ReadPage() gets delayed. And this timing shift somehow masks the handshake issue. 

May be I should create separate issue about the broken handshake. 

I totally agree with your original thinking. Delays after return Wire.requestFrom() have no sense and should be removed finally.

Koepel, 2016-01-20:
Clear.

The PN532 does sometimes not acknowledge its own slave address.
That means that a test for acknowledge and a test for ready should be added.
Perhaps something as is done here : https://github.com/adafruit/Adafruit_MPR121_Library/blob/master/Adafruit_MPR121.cpp
The Wire.requestFrom() is repeated until it returns the number of requested bytes. Maybe also a Wire.beginTransmission() and Wire.endTransmission() is enough, since the Wire.endTransmission returns an error if no acknowledge was received for the slave address.

After that, the first byte is a status byte with a RDY bit. It means that Wire.requestFrom() should be requested over and over again, until the RDY bit is 1.

Perhaps the both tests can be combined into one loop.

It is the same issue, I will change the subject.

mevst, 2016-01-25:
It looks like test for PN532 ready is wrongly excluded from sendCommandCheckAck() here:

  // For SPI only wait for the chip to be ready again.
  // This is unnecessary with I2C.
  if (_usingSPI) {
    if (!waitready(timeout)) {
      return false;
    }
  }

Actually the ready check must be done independently from I2C or SPI interface selection. It might be not needed for HSU (high speed UART) interface. The library does not support HSU anyway.

I removed the incorrect flagging around waitready() and left just this:
    if (!waitready(timeout)) {
      return false;
    }

And now it is possible to remove the unexplained delay() calls spread in the code. With this changes the time to dump the full content of 1K NTAG card dropped from 18 sec to 2 sec.

microbuilder, 2016-01-25:
If you have a pull request we'll be happy to review it.

mevst, 2016-08-24:
I've submitted pull request #26 for the issue fix.
Please review and test.

Koepel, 2016-08-24:
I looked at the code for the use of the Wire library, and that is okay.
The readdata() function reads more data, and I don't understand why (9 instead of 8, 22 instead of 20). I don't have the chip myself.

I wrote before that sometimes the PN532 does not acknowledge its own slave address, when it is busy. But I can't find that in the datasheet. Perhaps I was mistaken the low level I2C ACK with a higher level ACK frame.
What did NXP do ? The datasheet a "confidential document" ? [http://www.nxp.com/products/identification-and-security/nfc-and-reader-ics/nfc-controller-solutions/nfc-integrated-solution:PN5321A3HN](url)

mevst, 2016-08-24:
Thanks Koepel for the feedback,

readdata() should read all the bytes that PN532 wants to send towards host. Otherwise PN532 doesn't clear RDY handshake line. Actually last byte in the frame is a postamble byte, normally 0x00. 
I think that original developers assumed that it is possible to save a little time by skipping reading of the last byte with well known value. Or even skip reading of more bytes in the end of a data frame if their values are not used in the code then. However this breaks the handshake. To fix it I changed amount of bytes to read. Now this amount includes postamble byte.

It is possible to find old version of the datasheet (aka User Manual) from the Adafruit site. 
On page 42 there are some words about I2C acknowledge:
> Indeed, as the PN532 is configured as pure I2C slave, and as the host controller has the
possibility to discontinue the current process of the PN532 (as described in §6.2.2.1-d
and 6.2.2.2-b), the PN532 has to be configured in a mode such as it must acknowledge
the entire frame coming from the host controller (here acknowledge must be understood 
as the I2C meaning).
However, it may happen that the PN532 does not acknowledge its own address
immediately after having finished a previous exchange.

So far in my testing I haven't seen a case when PN532 does not respond with ACK to own I2C address. And the meaning of the full sentence is so fuzzy that I can not interpret it clearly. I could try to ask on NXP support forum while I am not sure that they will respond without signed NDA :)

Koepel, 2016-08-25:
Thanks for the explanation. You could add that as comment in the code.

caternuson, 2023-02-16:
Closing. The I2C/SPI (and now HSU) comms have been updated and delay removed.

closed 2023-02-16 by caternuson