adafruit/Adafruit-PN532 #20: esp8266

sfranzyshen, 2016-02-16:
Please add support for the esp8266 arduino platform

Found chip PN532
Firmware ver. 1.6
Waiting for an ISO14443A card

but nothing from the cards ...

glop102, 2016-02-16:
I believe I will get this working in about 2 weeks time. I am adding the breakout to the ESP for a door lock system. You can likely do it yourself using I2C if you dig into the library and make sure the library doing the I2C is properly set to work on the esp8266.

Apparently the Wire library will work but ONLY for master mode on the ESP. I am using the esp-12 model, are you working with this one as well?

glop102, 2016-02-27:
I have been playing with this chip on the ESP. I have replicated the behavior you have described several times but it seems inconsistent as to when it happens. I will reset the system and it then works fine. I believe it has something to do with wire length and timings, but i will look into it more on monday. I am using software SPI and so may be a bug with that. Also, please reply with what the hardware setup is that you are using.

sfranzyshen, 2016-02-27:
I am using a esp-12 with a itead pn532 card set to spi mode ... hooked up to the esp-12's hardware spi interface ...

glop102, 2016-02-28:
The module you are using seems to be a super simple breakout similar to what i am using. It says in the documentation that the chip can only go up to 5MHz (pg 45) and so perhaps the SPI library is not taking into account the faster speed of the ESP and so the speed of the SPI needs to be reduced. Try taking a 0 off the define for SPISettings(Speed,ByteType,Mode) on the speed argument.

As for the software SPI, i think i may have to add a delay in the write loop, just a couple microseconds, but enough to make it less that 5 MHz. Will check the software side on monday. Lets me hear your results.

future reference: the handbook for the pn532
http://www.nxp.com/documents/user_manual/141520.pdf

glop102, 2016-02-29:
upon more testing, i cant tell you what the problem is. The chip (pn532) seems to hang if it is not power cycled but you try to attach to it a second time with the library. I typically can get about 3-4 reads before it hangs if the card is already in place but then it stops working until it is powered off. Adding delays in the software SPI did nothing to help (though i did correct an ordering bug of the clock that was bugging me) and the hardware based SPI has the EXACT same behavior and so i doubt slowing it down will help at all. Let me know what your results are.

sfranzyshen, 2016-03-17:
I am bailing on the SPI interface (for now) I have the i2c working flawless with changes to the Seeed-Studio/PN532 driver ... see [here](https://github.com/Seeed-Studio/PN532/issues/44)

dilushan, 2017-02-24:
I faced also that hanging problem earlier. After that i made several changes and fixed it. So i cant tell what exactly made hang go away. Now its working flawlessly with software SPI.

mevst, 2017-02-24:
To @sfranzyshen 
I've added support for ESP8266 by pull request #26.
Could you test it?

dilushan, 2017-02-24:
Btw infinity loop in this code segments in examples result in a crash in esp8266. I know its not important. But when peoples see this right at the beginning lead to confusions. I know because i was scared.
```
 if (! versiondata) {
    Serial.print("Didn't find PN53x board");
    while (1); // halt
  }
```
simple solution is

```
  if (! versiondata) {
    Serial.print("Didn't find PN53x board");
    while (1) delay(100); // halt
  }

```

sleepdefic1t, 2017-09-29:
Not sure if it helps,
but I've got SPI working fine between a Feather Huzzah and PN532 using the following:

> #define PN532_SCK  (14)
> #define PN532_MOSI (13)
> #define PN532_MISO (12)
> #define PN532_SS  (5)
>
> Adafruit_PN532 nfc(PN532_SCK, PN532_MISO, PN532_MOSI, PN532_SS);

[image]

cjfradley, 2018-03-18:
@sleepdefic1t 
Hi Simon
So I have been able to get the ESP8266 to recognise the pn532 board.
But it won't read the tags
My Setup is as follows

```
#include "ArduinoJson.h"
#include <ESP8266WiFi.h>
#include <Adafruit_PN532.h>

const char* ssid     = "c&j's hotspot";
const char* password = "chrusimusi@home175";
    
#define PN532_SCK (14)
#define PN532_MOSI (13)
#define PN532_MISO (12)
#define PN532_SS (4)

Adafruit_PN532 nfc(PN532_SCK, PN532_MISO, PN532_MOSI, PN532_SS);

void setup() {
  
  Serial.begin(115200);
  //SPI.begin();
  delay(100);
 
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("Netmask: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("Gateway: ");
  Serial.println(WiFi.gatewayIP());

  Serial.println("");
  Serial.println("Starting nfc"); 
  nfc.begin();
  uint32_t versiondata = nfc.getFirmwareVersion();
  if (! versiondata) {
    Serial.print("Didn't find PN53x board");
    while (1); // halt
  }
  // Got ok data, print it out!
  Serial.print("Found chip PN5"); Serial.println((versiondata>>24) & 0xFF, HEX); 
  Serial.print("Firmware ver. "); Serial.print((versiondata>>16) & 0xFF, DEC); 
  Serial.print('.'); Serial.println((versiondata>>8) & 0xFF, DEC);

  // configure board to read RFID tags
  nfc.SAMConfig();
  
}

void loop() {
  uint8_t success;
  uint8_t uid[] = { 0, 0, 0, 0, 0, 0, 0 };  // Buffer to store the returned UID
  uint8_t uidLength;
  
  Serial.println();
  Serial.println("waiting...");
  Serial.println();
  success = nfc.readPassiveTargetID(PN532_MIFARE_ISO14443A, uid, &uidLength);
  if(success){
    Serial.println("found chip");
    delay(2000);
  }
}
```

So the setup works real nice.
But the success part in the loop just won't bite.
No tag is being seen.
Any Ideas what that might be ?

Thank you for any help

beicnet, 2020-06-25:
Hey @dilushan , @sfranzyshen did you ever try to use PN532 over SPI while executing GET with ESP8266HttpClient ?

caternuson, 2023-02-16:
Closing. Tested `readMifare` example on a Feather ESP8266 via I2C and works:

[image]

closed 2023-02-16 by caternuson