adafruit/Adafruit-PN532 #41: SPI mode not working

facetime88, 2018-01-09:
- Arduino board: Arduino DUE

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.5

I tried to use the example in SPI mode but it always said it didn't found the board. I tried before with adafruit nfc library with I2C mode and works well. But since I need use the SPI mode for p2p transfer, then I replace the adafruit with elechouse library, but still SPI mode not working.

jenschr, 2018-01-26:
@facetime88 if I2C works - did you change the SEL0 and SEL1 jumpers before trying SPI?

msoula, 2018-01-27:
I'm using the last version of this lib in SPI mode and I have no issue to report. Maybe you forgot something during init process.

Here are my init step (I've get rid of custom instructions):
```
// init controller
Adafruit_PN532 controller(ADAFRUIT_PN532_CLOCK_PIN, ADAFRUIT_PN532_MISO_PIN, ADAFRUIT_PN532_MOSI_PIN, ADAFRUIT_PN532_SS_PIN);

// don't know why but this seems necessary
digitalWrite(ADAFRUIT_PN532_SS_PIN, HIGH);

controller.begin();

// check controller
uint32_t version;
if (0 == (version = controller.getFirmwareVersion()))
{
    Serial.println("Didn't find PN53x board");
    return false;
}

// configure to read nfc tags
if (!controller.SAMConfig()) 
{
    Serial.println("Cannot configure the Secure Access Module");
    return false;
}

//[...]
```

I hope this will help you.

omersiar, 2018-03-21:
> // don't know why but this seems necessary
>digitalWrite(ADAFRUIT_PN532_SS_PIN, HIGH);

This is simply because this library does not care about other slaves on SPI bus. This must be fixed too

dinglewanker, 2018-04-27:
Yes.  it's working for me on the DUE board.  I'm using the hardware SPI pins and using pin 3 for CSEL.

// RFID Board pins
#define PN532_CSEL   3

// create rfi nfc board obj. uses hardward pins on DUE board. (in the middle of the board)
Adafruit_PN532 nfc(PN532_CSEL);

....
..

void initRFId() {
  nfc.begin();

  uint32_t versiondata = nfc.getFirmwareVersion();
  if (! versiondata) {
    lcd_clear_line(3); lcd.setCursor(0,3);
    lcd.print("RFID Init Fail");
  } else {
    lcd_clear_line(3); lcd.setCursor(0,3);
    lcd.print("RFID Init GOOD");
    // configure board to read RFID tags
    nfc.SAMConfig();
  }
}

tgikal, 2018-10-02:
I know it's probably a dead topic, but did you switch the dip switches/jumpers to SPI connection? Mine came defaulted to HSU and SPI did not work until I turned the SPI dip on.

vanquang741987, 2019-07-15:
HI @msoula: May you tell me how do you define SPI PIN for Arduino Due? For example what is ADAFRUIT_PN532_CLOCK_PIN in Arduino Due?ADAFRUIT_PN532_MISO_PIN? ADAFRUIT_PN532_MOSI_PIN?ADAFRUIT_PN532_SS_PIN?
Thanks

msoula, 2019-07-26:
On my case, I have:

/* AdaFruit PN532 */
#define ADAFRUIT_PN532_MISO_PIN     (MISO) /* MISO */
#define ADAFRUIT_PN532_MOSI_PIN     (MOSI) /* MOSI */
#define ADAFRUIT_PN532_CLOCK_PIN    (SCK)  /* SCK */
#define ADAFRUIT_PN532_SS_PIN                (13) /* PB7 */

You can read more about SPI on arduino on [this page](https://www.arduino.cc/en/Reference/SPI).

ladyada, 2020-02-24:
we refactored SPI, its probably fixed in 1.2.0+

closed 2020-02-24 by ladyada