adafruit/Adafruit-PN532 #48: Feather M0 spi

DaveCalaway, 2018-06-12:
- Arduino board:  Adafruit Feather M0 basic

- Arduino IDE version :  1.8.5

I use the example "readMifareClassic" in this library. 
I check the compatibility in the main page, i attached the device with SPI, but the compiler give me this error:
`
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp: In function 'void i2c_send(uint8_t)':
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:55:15: error: 'Wire1' was not declared in this scope
  #define WIRE Wire1
               ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:90:5: note: in expansion of macro 'WIRE'
     WIRE.write((uint8_t)x);
     ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp: In function 'uint8_t i2c_recv()':
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:55:15: error: 'Wire1' was not declared in this scope
  #define WIRE Wire1
               ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:104:12: note: in expansion of macro 'WIRE'
     return WIRE.read();
            ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp: In member function 'void Adafruit_PN532::begin()':
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:55:15: error: 'Wire1' was not declared in this scope
  #define WIRE Wire1
               ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:213:5: note: in expansion of macro 'WIRE'
     WIRE.begin();
     ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp: In member function 'void Adafruit_PN532::readdata(uint8_t*, uint8_t)':
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:55:15: error: 'Wire1' was not declared in this scope
  #define WIRE Wire1
               ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:1596:5: note: in expansion of macro 'WIRE'
     WIRE.requestFrom((uint8_t)PN532_I2C_ADDRESS, (uint8_t)(n+2));
     ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp: In member function 'void Adafruit_PN532::writecommand(uint8_t*, uint8_t)':
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:55:15: error: 'Wire1' was not declared in this scope
  #define WIRE Wire1
               ^
/Users/dave/Documents/Arduino/libraries/Adafruit_PN532/Adafruit_PN532.cpp:1697:5: note: in expansion of macro 'WIRE'
     WIRE.beginTransmission(PN532_I2C_ADDRESS);
`

BenBergman, 2019-12-06:
I had a similar issue and just commented out the `SerialUSB` line. After doing that the examples worked for me.

ladyada, 2020-02-24:
fixed in 1.2.0+

closed 2020-02-24 by ladyada