adafruit/Adafruit-PN532 #52: Each second passiv read of Target ID does not find a card.

weingaunity, 2018-09-10:
Hi, i'm using the readPassiveTargetID function.
Cards with 4byte-IDs are detected permanently. But each 2nd read of a card with 7 bytes does not work.

Any advice?

Thg.

Output @ 4byte Cards:
NOCARD
NOCARD
CARDFOUND   // placed card
CARDFOUND
CARDFOUND
CARDFOUND   // removed card
NOCARD
NOCARD

Output @ 7byte Cards:
NOCARD
NOCARD
CARDFOUND   // placed card
NOCARD
CARDFOUND
NOCARD
CARDFOUND
NOCARD
CARDFOUND   // removed card
NOCARD
NOCARD

popy2k14, 2021-01-12:
Late ... but hey beter than no answer :-)

I have exactly the same issue!!! Did'nt tested if it differs with UID lengths but i also get just an target back every second call to readPassiveTargetID() function.

I have found an "dirty" workaround, but not sure if i can go this way (lower performance because of calling the function twice...)
See my description/workaround here: https://community.nxp.com/t5/NFC/PN532-Unreliable-reading-of-2-tags-at-once/m-p/1210334/highlight/true#M8071

@weingaunity did you ever found an solution to this

caternuson, 2023-02-20:
Please provide a complete example sketch that can be used to demonstrate this issue for local testing.

caternuson, 2023-02-23:
Possibly related to #93

PockyBum522, 2023-06-27:
I was having this behavior. I enabled some of the debugs in the library, and saw there was NO ACK FRAME messages. 

I made changes in Adafruit_PN532.cpp:

Under line 335 where it says   // I2C TUNING
I changed this on line 336:

delay(SLOWDOWN); (Original line in 1.3.1)

to delay(1); (My modification.)

This means a delay of 1ms no matter what interface you're using. I'm using SPI. Before the modification, the library would only delay if you're using i2c. This fixed my issue.

I have board v1.3, I am using library 1.3.1. I see this behavior on two separate Adafruit PN532 boards. Both 1.3. This fixed the issue on both of them. I'm using a ESP32-S3-DevKitC-1 with both boards (Not the same one, this issue exists on two independent adafruit boards with a different ESP32-S3-DevKitC-1 on each.

The weirdest part is it WAS WORKING on one of the setups without the delay a few days ago. I know I haven't modified the library at all until today. I got nothin'

nilswinking, 2023-11-21:
I was having problems reading data from a Mifare ultralight tag and setting the delay to 1 fixed it.

caternuson, 2023-11-21:
Is everyone experiencing this issue using SPI?

nilswinking, 2023-11-21:
I was using SPI and for I2C the delay is set to 1 anyway

caternuson, 2023-11-21:
Correct. Asking as still looking for an example sketch and/or information about how to recreate this locally.

nilswinking, 2023-11-21:
I just used the ntag2xx_read example provided by this library and then changed the delay to 1. Before I got no ack frames. Interestingly reading the ID worked fine, but even when commenting out that line and just skipping to reading the tag directly it didn't work. But with debug enabled it didn't get the error in this part of the code
```C++
  if (!sendCommandCheckAck(pn532_packetbuffer, 4)) {
#ifdef MIFAREDEBUG
    PN532DEBUGPRINT.println(F("Failed to receive ACK for write command"));
#endif
    return 0;
  }
```

caternuson, 2023-11-22:
Still can't recreate this locally. Running the `ntag2xx_read` on a Metro M4 using same pins defined in example (software SPI), it runs without issue for multiple reads:

```
08:32:35.837 -> Send a character to scan another tag!
08:32:37.791 -> Found an ISO14443A card
08:32:37.791 ->   UID Length: 7 bytes
08:32:37.791 ->   UID Value: 0x04 0xFA 0x8C 0x42 0xD9 0x11 0x90
08:32:37.791 -> 
08:32:37.791 -> Seems to be an NTAG2xx tag (7 byte UID)
08:32:37.791 -> PAGE 00: Unable to read the requested page!
08:32:37.827 -> PAGE 01: 42 D9 11 90  B⸮.⸮
08:32:37.827 -> PAGE 02: 1A 48 00 00  .H..
08:32:37.857 -> PAGE 03: E1 10 12 00  ⸮...
08:32:37.890 -> PAGE 04: 01 03 A0 0C  ..⸮.
08:32:37.890 -> PAGE 05: 34 03 00 FE  4..⸮
08:32:37.923 -> PAGE 06: 00 00 00 00  ....
08:32:37.956 -> PAGE 07: 00 00 00 00  ....
08:32:37.956 -> PAGE 08: 00 00 00 00  ....
08:32:37.989 -> PAGE 09: 00 00 00 00  ....
08:32:37.989 -> PAGE 10: 00 00 00 00  ....
08:32:38.023 -> PAGE 11: 00 00 00 00  ....
08:32:38.056 -> PAGE 12: 00 00 00 00  ....
08:32:38.056 -> PAGE 13: 00 00 00 00  ....
08:32:38.089 -> PAGE 14: 00 00 00 00  ....
08:32:38.122 -> PAGE 15: 00 00 00 00  ....
08:32:38.122 -> PAGE 16: 00 00 00 00  ....
08:32:38.155 -> PAGE 17: 00 00 00 00  ....
08:32:38.155 -> PAGE 18: 00 00 00 00  ....
08:32:38.188 -> PAGE 19: 00 00 00 00  ....
08:32:38.221 -> PAGE 20: 00 00 00 00  ....
08:32:38.221 -> PAGE 21: 00 00 00 00  ....
08:32:38.254 -> PAGE 22: 00 00 00 00  ....
08:32:38.287 -> PAGE 23: 00 00 00 00  ....
08:32:38.287 -> PAGE 24: 00 00 00 00  ....
08:32:38.321 -> PAGE 25: 00 00 00 00  ....
08:32:38.321 -> PAGE 26: 00 00 00 00  ....
08:32:38.354 -> PAGE 27: 00 00 00 00  ....
08:32:38.387 -> PAGE 28: 00 00 00 00  ....
08:32:38.387 -> PAGE 29: 00 00 00 00  ....
08:32:38.420 -> PAGE 30: 00 00 00 00  ....
08:32:38.420 -> PAGE 31: 00 00 00 00  ....
08:32:38.453 -> PAGE 32: 00 00 00 00  ....
08:32:38.486 -> PAGE 33: 00 00 00 00  ....
08:32:38.486 -> PAGE 34: 00 00 00 00  ....
08:32:38.519 -> PAGE 35: 00 00 00 00  ....
08:32:38.552 -> PAGE 36: 00 00 00 00  ....
08:32:38.552 -> PAGE 37: 00 00 00 00  ....
08:32:38.586 -> PAGE 38: 00 00 00 00  ....
08:32:38.586 -> PAGE 39: 00 00 00 00  ....
08:32:38.619 -> PAGE 40: 00 00 00 BD  ...⸮
08:32:38.652 -> PAGE 41: 04 00 00 FF  ...⸮
08:32:38.652 -> 
08:32:38.652 -> 
08:32:38.652 -> Send a character to scan another tag!
08:32:40.540 -> Found an ISO14443A card
08:32:40.540 ->   UID Length: 7 bytes
08:32:40.540 ->   UID Value: 0x04 0xFA 0x8C 0x42 0xD9 0x11 0x90
08:32:40.540 -> 
08:32:40.540 -> Seems to be an NTAG2xx tag (7 byte UID)
08:32:40.540 -> PAGE 00: Unable to read the requested page!
08:32:40.540 -> PAGE 01: 42 D9 11 90  B⸮.⸮
08:32:40.573 -> PAGE 02: 1A 48 00 00  .H..
08:32:40.606 -> PAGE 03: E1 10 12 00  ⸮...
08:32:40.606 -> PAGE 04: 01 03 A0 0C  ..⸮.
08:32:40.639 -> PAGE 05: 34 03 00 FE  4..⸮
08:32:40.672 -> PAGE 06: 00 00 00 00  ....
08:32:40.672 -> PAGE 07: 00 00 00 00  ....
08:32:40.705 -> PAGE 08: 00 00 00 00  ....
08:32:40.705 -> PAGE 09: 00 00 00 00  ....
08:32:40.739 -> PAGE 10: 00 00 00 00  ....
08:32:40.772 -> PAGE 11: 00 00 00 00  ....
08:32:40.772 -> PAGE 12: 00 00 00 00  ....
08:32:40.805 -> PAGE 13: 00 00 00 00  ....
08:32:40.838 -> PAGE 14: 00 00 00 00  ....
08:32:40.838 -> PAGE 15: 00 00 00 00  ....
08:32:40.871 -> PAGE 16: 00 00 00 00  ....
08:32:40.871 -> PAGE 17: 00 00 00 00  ....
08:32:40.904 -> PAGE 18: 00 00 00 00  ....
08:32:40.937 -> PAGE 19: 00 00 00 00  ....
08:32:40.937 -> PAGE 20: 00 00 00 00  ....
08:32:41.000 -> PAGE 21: 00 00 00 00  ....
08:32:41.000 -> PAGE 22: 00 00 00 00  ....
08:32:41.024 -> PAGE 23: 00 00 00 00  ....
08:32:41.024 -> PAGE 24: 00 00 00 00  ....
08:32:41.047 -> PAGE 25: 00 00 00 00  ....
08:32:41.070 -> PAGE 26: 00 00 00 00  ....
08:32:41.103 -> PAGE 27: 00 00 00 00  ....
08:32:41.103 -> PAGE 28: 00 00 00 00  ....
08:32:41.136 -> PAGE 29: 00 00 00 00  ....
08:32:41.136 -> PAGE 30: 00 00 00 00  ....
08:32:41.169 -> PAGE 31: 00 00 00 00  ....
08:32:41.202 -> PAGE 32: 00 00 00 00  ....
08:32:41.202 -> PAGE 33: 00 00 00 00  ....
08:32:41.235 -> PAGE 34: 00 00 00 00  ....
08:32:41.268 -> PAGE 35: 00 00 00 00  ....
08:32:41.268 -> PAGE 36: 00 00 00 00  ....
08:32:41.301 -> PAGE 37: 00 00 00 00  ....
08:32:41.301 -> PAGE 38: 00 00 00 00  ....
08:32:41.334 -> PAGE 39: 00 00 00 00  ....
08:32:41.367 -> PAGE 40: 00 00 00 BD  ...⸮
08:32:41.367 -> PAGE 41: 04 00 00 FF  ...⸮
08:32:41.367 -> 
08:32:41.367 -> 
08:32:41.367 -> Send a character to scan another tag!
```

pcb-ninja, 2023-12-11:
Same issue on Particle Xenon. Delay(1) fixed the issue. Using the ntag2xx_read example.