adafruit/Adafruit-PN532 #53: SPI.beginTransaction is called twice, locks up on ESP32

robbi5, 2018-09-25:
The `begin` function blocks itself, when `SPI_HAS_TRANSACTION` is defined and Hardware SPI is used.

In `begin` `SPI.beginTransaction` is called first at 
https://github.com/adafruit/Adafruit-PN532/blob/941cac4d6b77c4a5ea8ba10e416be5b930b4ffea/Adafruit_PN532.cpp#L197
then `sendCommandCheckAck` is called which itself calles `writecommand` which tries to start another transaction with `SPI.beginTransaction`.
https://github.com/adafruit/Adafruit-PN532/blob/941cac4d6b77c4a5ea8ba10e416be5b930b4ffea/Adafruit_PN532.cpp#L1649

On an ESP32 with the arduino-esp32 framework the [`SPI.beginTransaction`](https://github.com/espressif/arduino-esp32/blob/e5ea089a7fd5cc8a96d68aae64dc3b4dcef2315e/libraries/SPI/src/SPI.cpp#L127) calls the `spiTransaction` function which uses an `SPI_MUTEX_LOCK`:

https://github.com/espressif/arduino-esp32/blob/1.0.0/cores/esp32/esp32-hal-spi.c#L702-L707 

This combination of calling `SPI.beginTransaction` twice (in `begin` and `writecommand`) causes a hang on the second invocation, because the `SPI_MUTEX_LOCK` is never unlocked.

----

- Arduino board:  ESP32 w/ arduino-esp32 (tag: 1.0.0)

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

```
#define SPI_HAS_TRANSACTION
#include <SPI.h>
#define PN532DEBUG
#include <Adafruit_PN532.h>

Adafruit_PN532 nfc(18);

void setup() {
  nfc.begin(); // see debug output, hangs at "Sending:" (no bytes after this message)
}
void loop() {
}
```

ladyada, 2018-09-25:
oof - can you try removing it from begin() does it work now?

robbi5, 2018-09-25:
yep, then it doesn't hang anymore. now we're still figuring out why a timeout happens, but the locking problem is gone :)

izedgo, 2020-02-10:
Why a PR is not accepted yet? I spent a lot of time to fix this :(

factor8, 2020-02-20:
We are also running into this problem. Can we get @izedgo 's PR merged soon?

ladyada, 2020-02-24:
fixed in 1.2.0+

closed 2020-02-24 by ladyada