adafruit/Adafruit-PN532 #55: NDEF NTAG215 and 216 compatability

tgikal, 2018-10-01:
The nfc.ntag2xx_WriteNDEFURI(ndefprefix, url, dataLength) is written with dataLength defined as a uint8_t, which limits the data size to 256, even though NTAG215 and NTAG216 tags have much more memory than this.

tgikal, 2018-10-01:
Managed to edit the .cpp and .h files to accommodate this issue, since past experience has proven Adafruit does not pay attention to issues and will not correct this.

The only other change made is I removed the annoying and pointless "TIMEOUT!" message from waitready.

[Adafruit-PN532.zip](https://github.com/adafruit/Adafruit-PN532/files/2435393/Adafruit-PN532.zip)
You'll probably have to remove any other versions to make the IDE choose the correct one.

ladyada, 2018-10-01:
hi @tgikal please stay positive and constructive in comments, we have 1000+ repos to maintain - not every issue & PR is handled immediately. please submit a PR if you have tested the code!

tgikal, 2018-10-01:
There are 19 pending pull requests, 12 of which are over 1 year old, one of which already addressed the "TIMEOUT!" message.

caternuson, 2023-02-20:
Looks like a dupe of #38