adafruit/Adafruit-PN532 #69: Library no longer compiles on platformio for ESP32

mbariola, 2019-12-30:
- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

Esp32 Dev Module w/SSD1306 Screen

- Arduino IDE version (found in Arduino -> About Arduino menu):

PlarformIO for VS Code, Home v3.0.1 Core v4.1.0

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

I have compiled this sketch successfully many times up to a couple month ago when I last used it. Then I tried recompiling and got these errors (and many warnings too, but the warnings have always been there - the errors forbidding me to compile is something new):

```
In file included from .pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:60:0:
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h: In constructor 'Adafruit_PN532::Adafruit_PN532(uint8_t, uint8_t, uint8_t, uint8_t)':
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:29: error: 'Adafruit_PN532::_miso' will be initialized after [-Werror=reorder]   
   uint8_t _ss, _clk, _mosi, _miso;
                             ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:22: error:   'uint8_t Adafruit_PN532::_mosi' [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
                      ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:127:1: error:   when initialized here [-Werror=reorder]
 Adafruit_PN532::Adafruit_PN532(uint8_t clk, uint8_t miso, uint8_t mosi, uint8_t ss):
 ^
In file included from .pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:60:0:
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:22: error: 'Adafruit_PN532::_mosi' will be initialized after [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
                      ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:11: error:   'uint8_t Adafruit_PN532::_ss' [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
           ^
Compiling C:\Users\Massi\AppData\Local\Temp\pio-build\CROVE_RFID_Test-7a3db15896\esp32dev\lib142\BLE\BLEBeacon.cpp.o
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:127:1: error:   when initialized here [-Werror=reorder]
 Adafruit_PN532::Adafruit_PN532(uint8_t clk, uint8_t miso, uint8_t mosi, uint8_t ss):
 ^
In file included from .pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:60:0:
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h: In constructor 'Adafruit_PN532::Adafruit_PN532(uint8_t, uint8_t)':
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:29: error: 'Adafruit_PN532::_miso' will be initialized after [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
                             ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:22: error:   'uint8_t Adafruit_PN532::_mosi' [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
                      ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:152:1: error:   when initialized here [-Werror=reorder]
 Adafruit_PN532::Adafruit_PN532(uint8_t irq, uint8_t reset):
 ^
In file included from .pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:60:0:
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:22: error: 'Adafruit_PN532::_mosi' will be initialized after [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
                      ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:11: error:   'uint8_t Adafruit_PN532::_ss' [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
           ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:152:1: error:   when initialized here [-Werror=reorder]
 Adafruit_PN532::Adafruit_PN532(uint8_t irq, uint8_t reset):
 ^
In file included from .pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:60:0:
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h: In constructor 'Adafruit_PN532::Adafruit_PN532(uint8_t)':
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:29: error: 'Adafruit_PN532::_miso' will be initialized after [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
                             ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:22: error:   'uint8_t Adafruit_PN532::_mosi' [-Werror=reorder]
   uint8_t _ss, _clk, _mosi, _miso;
                      ^
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:173:1: error:   when initialized here [-Werror=reorder]
 Adafruit_PN532::Adafruit_PN532(uint8_t ss):
 ^
In file included from .pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.cpp:60:0:
.pio\libdeps\esp32dev\Adafruit PN532_ID29\Adafruit_PN532.h:202:22: error: 'Adafruit_PN532::_mosi' will be initialized after [-Werror=reorder]
[... 228 lines cut ...]
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
build_dir = ${sysenv.TEMP}/pio-build/$PROJECT_HASH

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
  Wire
  Adafruit PN532
  ESP8266_SSD1306
  SPI
monitor_speed = 115200
board_build.f_cpu = 80000000L
```

I want to stress that the sketch compiled fine up till now (it is running on a board, I just wanted to change a line but it will no longer successfully compile). It seems a problem of compiler options here but am not sure...

Any help or pointer in the right direction very much appreciated as am currently stuck.

mbariola, 2020-01-02:
As suggested by @manuelbl [here:](https://community.platformio.org/t/error-adafruit-pn532-miso-will-be-initialized-after-werror-reorder/11275/3)

changing line 202 in Adafruit_PN532.h to 
`  uint8_t _clk, _miso, _mosi, _ss;`

provides a band-aid to the problem. Still it might be of interest investigating why the problem occurs (or if it breaks other environments) and provide a more robust fix

ladyada, 2020-02-24:
fixed in 1.2.0+

closed 2020-02-24 by ladyada