adafruit/Adafruit-PN532 #73: Timeout-Error when reading ISO1443 with long APDU

asdfjkl, 2020-02-28:
- Arduino board:  Arduino Due
- Arduino IDE version 1.8.12
- NFC shield connected via I2C

- Take a ISO14443 card (A, but patching the code to run with type B card will result in the same problem). For example: An NXP JCOP card (P5, P60), G&D Smartcafe JavaCard, or any other JavaCard. Or a German eID card.

- If we send a short APDU (in my setup <= 23 byte long), the card reply is correctly read. Sending an APDU should result in an ISO7816 status code, e.g. sending a valid APDU should result in 0x9000, and sending an invalid APDU should result in an error code like 0x6700 or 0x6881.

- If we send a longer APDU (in my setup >= 24 byte), this results in a timeout.

More precisely, I was able to debug this down to:

calling 
    inDataExchange(apdu, apdu_len, response, &response_len);
-> if (!sendCommandCheckAck(pn532_packetbuffer,sendLength+2,1000)) will result in false returned.

Here within 
-> sendCommandCheckAck the call will fail at
   // Wait for chip to say its ready!
   if (!waitready(timeout)) {

Increasing the timeout - even to large values - does not help, so I suspect some other fundamental (timing-related?) issue...

Tested this also with different versions of this library, i.e. "official 1.4" as well as the current branch, and the behaviour is the same.

asdfjkl, 2020-11-16:
more info: This seems to only affect I2C communication (both on 1.04 as well as 1.2.x). (Software) SPI communication works with longer APDUs.

alec87, 2020-12-20:
This is an error in the standard Arduino Libraries "twi.h and wire.h". I fixed it by changing the TWI_BUFFER_LENGTH in twi.h and the BUFFER_LENGTH in wire.h both from 32 to 64 (in my case, you may need more).

caternuson, 2023-02-20:
Closing. Possibly either a DUE + I2C quirk or fixed by library update to BusIO.

closed 2023-02-20 by caternuson