adafruit/Adafruit-PN532 #74: Added a new unknown dependency: #include <Adafruit_SPIDevice.h>

hvegh, 2020-03-05:
Build result:
.pio/libdeps/lolin32/Adafruit PN532_ID29/Adafruit_PN532.h:34:32: fatal error: Adafruit_SPIDevice.h: No such file or directory

hvegh, 2020-03-05:
Ok need to add: Adafruit_BusIO as an additional dependency. My bad..

ladyada, 2020-03-05:
it should be automatically installed for you, as it is a dependant library:
https://github.com/adafruit/Adafruit-PN532/blob/master/library.properties#L10
@ivankravets is the `depends` in library.properties parsed?

closed 2020-03-05 by hvegh

hvegh, 2020-03-06:
> it should be automatically installed for you, as it is a dependant library:
> https://github.com/adafruit/Adafruit-PN532/blob/master/library.properties#L10
> @ivankravets is the `depends` in library.properties parsed?

Thanks for the feedback. BTW I am using platformio...

ladyada, 2020-03-06:
@hvegh thanks - can you open an issue with platformio? it should automatically be installing the dependencies for you as specified :)

ivankravets, 2020-03-07:
Hi @ladyada and @hvegh 

Sorry for the delay. This is an issue on our side and we have just fixed it. @hvegh please re-run in PlatformIO IDE Terminal `pio upgrade --dev`. The final PIO Core 4.2.2 is going to be released next week.

ladyada, 2020-03-07:
yay! thank you so much @ivankravets :) please note that dependencies should be handled recursively if they are not. we are slowly converting all our old libraries to actions and also adding deps. if you see libraries that are missing deps please open an issue on that repo.

ivankravets, 2020-03-07:
> lease note that dependencies should be handled recursively

Yes, we handle them recursively. The problem was with our legacy library registry and we added a small hook to PIO Core to avoid this issue. The new universal registry for libraries, dev-platforms, toolchains is going to be released this month.

P.S: Still dream about a "Learn PlatformIO" section at https://learn.adafruit.com/ :) I saw the latest news about Zephyr RTOS & Adafruit. PlatformIO could provide a unified development environment for all Adafruit products where developers can easily switch between frameworks (Arduino, Zephyr, ESP-IDF) and use the same favorite dev-environment. Will it happen ever? 😊

ptorrone, 2020-03-07:
@ivankravets hi hi!

re:

> Still dream about a "Learn PlatformIO" section at https://learn.adafruit.com/ :) I saw the latest news about Zephyr RTOS & Adafruit. PlatformIO could provide a unified development environment for [...]

did you see the last email from awhile ago we sent over? we'd very much like a guide on learn.adafruit.com if you are up for it!

it was from last year, there was no reply though (July 21, 2019 at 12:52:52 PM EDT, Re: Hi from PlatformIO :))

ivankravets, 2020-03-07:
Hi @ptorrone,

It seems that I missed that letter but I remember it :( My fault. Sorry.

> we'd very much like a guide on learn.adafruit.com if you are up for it!

Do you have any hints about what could be interested in your audience? Do you have requirements for the format of an article? Does it make sense to start from Zephyr ROTS? We have full support in PlatformIO for Zephyr.

ptorrone, 2020-03-07:
@ivankravets all good, no worries. it can be Zephyr if you like, we think there will be more people looking for how to use Zephyr quickly and easily :) reply to that note and we'll set ya up on learn.adafruit.com !

ivankravets, 2020-03-07:
How can we post similar to https://learn.adafruit.com/blinking-led-with-zephyr-rtos/? Do you have "learn" dashboard for authors?
This guide is very complex for the typical developers and requires different steps depending on the operating system.

Please ping me at me@ikravets.com, let's finally have this guide.

ptorrone, 2020-03-07:
@ivankravets yep! we have an excellent learning system for authors, i replied to the email from last year we sent ya, look for it and reply back when ya can!

Neutral854, 2020-12-11:
> Build result:
> .pio/libdeps/lolin32/Adafruit PN532_ID29/Adafruit_PN532.h:34:32: fatal error: Adafruit_SPIDevice.h: No such file or directory

I had to specifically install this library to make it work:
https://github.com/adafruit/Adafruit_BusIO