adafruit/Adafruit-PN532 #76: Increasing the answer/command buffer size

salmg, 2020-04-07:
Noticed that the buffer used for PN532 communication is limited by 64 bytes:

`#define PN532_PACKBUFFSIZ 64`

- The C-APDU command length can be up to 261 bytes (CLA-INS-P1-P2-P3-255 data bytes-Le)
- The R-APDU returned to the host controller can have a length of 258 bytes (256 data bytes-SW1-SW2)

By design, the library uses just one buffer for commands and answers with fixed size(PN532_PACKBUFFSIZ) but a very low one.

Using only 64 bytes from the PN532 buffer is almost four times less its limit. This limitation affects directly the ISO/14443A or raw communication projects.

Making some tests, I noticed that the actual library design could be implemented with a buffer size of 255 without changing anything else:

`#define PN532_PACKBUFFSIZ 255`

asdfjkl, 2020-10-20:
can you confirm/deny that _sending_ a large apdu (say 80 bytes) is possible? In my setup ( https://github.com/adafruit/Adafruit-PN532/issues/73 ) increasing the command buffer for _receiving_ apdus works fine (large apdu replies are received ok), but _sending_ large apdus fails.

salmg, 2020-10-26:
I have to mentioned that I tested that buffer size using only the SPI configuration.

caternuson, 2023-02-20:
Looks similar to #38

mhaberler, 2023-12-09:
> can you confirm/deny that _sending_ a large apdu (say 80 bytes) is possible? In my setup ( #73 ) increasing the command buffer for _receiving_ apdus works fine (large apdu replies are received ok), but _sending_ large apdus fails.

see the manual reference in https://github.com/adafruit/Adafruit-PN532/issues/124 which is unambiguous