adafruit/Adafruit-PN532 #78: Apple Pay

giorgio80, 2020-06-11:
hello, 

I noticed that when passing my phone near the reader , Apple Pay automatically activate , even if there's no payment to do.   

It is possible to configure the card to don't make Apple Pay activate?

dherrada, 2020-06-11:
I could be wrong, but I'm pretty certain that this is on your phone's end. Barring disabling it on your phone, or having Apple make a change to how Apple Pay works, I don't believe there is a way to do so.

Basically Apple Pay will activate whenever the iPhone detects that it is near an NFC terminal (and often an RFID scanner, since they're pretty similar). I came across [this](https://discussions.apple.com/thread/6659565) thread, which was 6 years ago, and given that you're having what appears to be more or less the same issue (in that case it confused an RFID scanner with an NFC terminal, whereas since the PN532 does both, who knows which one it's detecting) it seems unlikely Apple plans on changing that any time soon.

closed 2020-06-12 by siddacious

giorgio80, 2020-06-12:
I'm maybe wrong , but could not the reader be configured to read raid but not NFC?

dherrada, 2020-06-12:
It will trigger in both instances. Apple pay works with NFC, but NFC and RFID are so similar that even if you only enabled RFID, it would still trigger apple pay.