adafruit/Adafruit-PN532 #92: SPI Wiring Not working on d1 mini

6000850, 2021-05-15:
Hello
This wiring is SPI, but after uploading the ReadTag.ino code, the message "Didn't find PN53x board" is displayed.

My Board : 
`Wemos D1 Mini`

My Wiring :
```
SCK => D5
MISO => D6
MOSI => D7
SS => D3
VCC => 3.3V
GND => GND
```
**When I connect the SS pin to D8. Upload gives an error and the code is not uploaded**

ReadTag.ino

```
#include <NfcAdapter.h>
#include <PN532/PN532/PN532.h>

    #include <SPI.h>
    #include <PN532/PN532_SPI/PN532_SPI.h>
    PN532_SPI pn532spi(SPI, 0);
    NfcAdapter nfc = NfcAdapter(pn532spi);

void setup(void) {
    SERIAL.begin(9600);
    SERIAL.println("NDEF Reader");
    nfc.begin();
}

void loop(void) {
    SERIAL.println("\nScan a NFC tag\n");
    if (nfc.tagPresent()) {
        NfcTag tag = nfc.read();
        tag.print();
    }
    delay(5000);
}
```

6000850, 2021-05-19:
SoftwareSerial worked! but SPI NOT Working!! why!?

IntegrityKnight, 2021-10-18:
I am use this PN532 module and WeMos D1 mini:
[https://nl.aliexpress.com/item/33014782756.html](https://nl.aliexpress.com/item/33014782756.html)

1. Hardware switch
Change to SPI mode

2. Wiring
```
SCK => D5
MISO => D6
MOSI => D7      
SS => D3      
VCC => 5V      
GND => GND
```

3. Coding
Use the Example [iso14443a_uid](https://github.com/adafruit/Adafruit-PN532/tree/master/examples/iso14443a_uid). And change the following code
```
// If using the breakout with SPI, define the pins for SPI communication.
#define PN532_SCK  (2)
#define PN532_MOSI (3)
#define PN532_SS   (D3)   //change to D3
#define PN532_MISO (5)
```
Comment "Adafruit_PN532 nfc(PN532_SCK, PN532_MISO, PN532_MOSI, PN532_SS);"
```
// Use this line for a breakout with a SPI connection:
// Adafruit_PN532 nfc(PN532_SCK, PN532_MISO, PN532_MOSI, PN532_SS);  //comment this line
```

Uncomment "Adafruit_PN532 nfc(PN532_SS);"

```
// Use this line for a breakout with a hardware SPI connection.  Note that
// the PN532 SCK, MOSI, and MISO pins need to be connected to the Arduino's
// hardware SPI SCK, MOSI, and MISO pins.  On an Arduino Uno these are
// SCK = 13, MOSI = 11, MISO = 12.  The SS line can be any digital IO pin.
Adafruit_PN532 nfc(PN532_SS); //uncomment this line
```

Enjoy!!

caternuson, 2023-02-16:
Closing. Looks like a different library is being used. There is no `PN532_SPI` class in this library.
```cpp
    PN532_SPI pn532spi(SPI, 0);
```

closed 2023-02-16 by caternuson