adafruit/Adafruit-PN532 #93: Mifare classic 7-byte UID authentication

hitriy, 2021-06-07:
Hello.
I'm trying to read data from Micare Classic card with 7-byte UID. 
I'm using mifareclassic_memdump as reference. 
To get data from card I've added new condition to length check
`
if (uidLength == 4 || uidLength == 7)
`
However this doesn't work.
According to the [documentation ](https://www.nxp.com/docs/en/application-note/AN10927.pdf) I must use byte 4,5,6,7 of the 7-byte UID, but still no luck:
```
Waiting for an ISO14443A Card ...
Sending : 0x0, 0x0, 0xFF, 0x4, 0xFC, 0xD4, 0x4A, 0x1, 0x0, 0xE1, 0x0,
Reading:  0x0 0x0 0xFF 0xF 0xF1 0xD5 0x4B 0x1 0x1 0x0 0x42 0x18 0x7 0x4 0x5D 0x46 0xB2 0x50 0x5D 0x80
Found 1 tags
ATQA: 0x42
SAK: 0x18
UID: 0x4 0x5D 0x46 0xB2 0x50 0x5D 0x80
Seems to be a Mifare Classic card (
4 byte UID)------------------------Sector 4-------------------------
Trying to authenticate card Using authentication KEY B: Sending : 0x0, 0x0, 0xFF, 0xF, 0xF1, 0xD4, 0x40, 0x1, 0x61, 0x10, 0x**, 0x**, 0x**, 0x**, 0x**, 0x**, 0xB2, 0x50, 0x5D, 0x80, 0x9C, 0x0, 
No ACK frame received!
Authentication error
Block 16 unable to authenticate
```

SebastiaanMerckx, 2021-06-17:
I'm also having issues with authentication of mifare classic 7-byte uid.
I do get an ACK however.
Do you wait for interrupt or work polling based?
You use command 0x61 (MIFARE_CMD_AUTH_B), have you tried AUTH_A?
And are you sure that block 0x10 is in valid range? How about block 0 or 4?

My current issue (using the last 4 bytes of UID, that was new information for me), is that I get 
0x00 0x00 0xFD 0x03 0xFD 0xD5 0x41 0x20
=> so seems like error code 0x20.

hitriy, 2021-06-17:
I've managed to get data from block 0x10, and yes, since it's mifare classic 4k block 16 is within addresses range. 
Sending last four bytes of UID works fine. 
I've used some direct editing, however it would be nice to have 7 and 10 bytes UID cards out of the box

SebastiaanMerckx, 2021-06-18:
Okay I was using an incorrect authentication key, I now got it working as well.
The library itself can be easily adapted (even though the table in the documentation doesn't mention 10-bytes UID?)

I can do the code change byt I'm not using an arduino, hence not really the direct library. If you are, I can send the code change for validation?

mkcoldwolf, 2024-10-07:
Thanks for informations! It helped
There is actually two issues

1. No ACK frame
This is due to the low-level hardware communication issue mentioned in #52 
https://github.com/adafruit/Adafruit-PN532/issues/52#issuecomment-1610001146
Change `uint8_t SLOWDOWN = 0;` to `uint8_t SLOWDOWN = 1;` resolve the issue

2. The library did not implement the 7 byte auth mechanism correctly
According to documents, it should send only the last 4 byte for auth frame, and frame length should kept unchanged,
but the library just tryna put all 7 byte into frame in method `mifareclassic_AuthenticateBlock`
Replace the for loop and the frame send with below:
```c++
for (i = 3; i < _uidLen; i++) {
    pn532_packetbuffer[10 + i - 3] = _uid[i]; /* 4 byte card ID */
  }

  if (!sendCommandCheckAck(pn532_packetbuffer, 14))
    return 0;
```