adafruit/Adafruit-PN532 #101: ESP32 I2C sharing was working fine

FStefanni, 2022-03-29:
Hi,

I was unaware of the requirement of unshared I2C on ESP32, so I wrongly reported an issue
to another Adafruit library ([here it is](https://github.com/adafruit/Adafruit_SSD1306/issues/238)).

The strange fact is that with the Arduino-ESP32 board version 2.0.0 it *was working fine*!
When upgrading to 2.0.2, it stops working.

So I am wondering what is going on... can someone enlighten me please?

Regards

caternuson, 2023-02-20:
Are you seeing this issue running one of the examples from this library?

Just recently test I2C on a Feather ESP32 V2 with BSP 2.0.6 and it worked fine.

Try updating ESP32 BSP to latest and verify things using one of the example from this library.

FStefanni, 2023-02-21:
Hi,

sorry, but I have no way to test it now.
Basically, it was more a try to understand why, let's say I was curious.
But as said, in the end we have used two I2C buses.

For what it can matter, this is what I found just by using various bsp and libraries:

* bsp 2.0.0: everything was working fine, even with unshared bus. Maybe there could have been already the requirement of unshared bus, but I never noticed malfunctions
* bsp 2.0.1: nothing was working -- this has been a buggy bsp release
* bsp 2.0.2/2.0.3/2.0.5: something better
* bsp 2.0.5: works fine, but respecting the unshared I2C requirement (never tested again unshared I2C)
* bsp 2.0.6: not tested

Except version 2.0.1 which was broken, I do not know if there were also issues on libraries side, which have been fixed meanwhile (so maybe also 2.0.2 was fine and the issue maybe was on the libraries). At the moment in production we use 2.0.5 with almost up to date libraries.

I am closing this issue since probably no more relevant for this library

Regards

closed 2023-02-21 by FStefanni