adafruit/Adafruit-PN532 #104: Android HCE random ID

braysonjohn148, 2022-10-09:
according to Flutter_NFC library: NfcEmulator.startNfcEmulator("00A40400F0010203040506","cd22c716","79e64d05ed6475dcd506b") cardAid, cardUid, [ aesKey] respectively. This emulates the card but gives a random ID on each tap. How do I know the user/tap that has tapped?

caternuson, 2023-02-20:
What is the hardware being used as the NFC target?

What sketch is being run for the PN532?

skibz, 2023-11-20:
Smartphones don't use static UID values for emulated cards as a security measure.

You have to implement your own [APDU](https://developer.android.com/reference/android/nfc/cardemulation/HostApduService) communication scheme between your Android app and microcontroller firmware.