adafruit/Adafruit-PN532 #113: I2C PN532 on Pi Pico doesnt work updating to 1.3.0

AnotherMatt92, 2023-03-12:
Board: Pi Pico running Arduino code
Pinout: I2C0 default (pins GP4/5)
Library version: 1.3.0
Code: Any library example using I2C rather than SPI

All examples upload and own code work when using V1.2.2 of this library, however updating this library to V1.3.0 stops it from working anymore
I have tried passing Wire as an additional argument but I couldnt get anywhere with it.
Reverting library to 1.2.2 gets the PN532 working again

caternuson, 2023-03-13:
Which specific example sketch are you running and what is being output in the Arduino Serial Monitor?

danyocom, 2023-03-15:
I am seeing the same issue (example does not work, board commands seem do not work other than dumping the chipid/version) with library version 1.3.0 on an ItsyBitsy nRF52840 Express.  When running the iso14443a_uid example the output is the following

Hello!
Found chip PN532
Firmware ver. 1.6

with library 1.2.3 cards are detected and the output is 

Hello!
Found chip PN532
Firmware ver. 1.6
Waiting for an ISO14443A card

caternuson, 2023-03-15:
@danyocom Since you're using a different platform (nrf), it may be a different issue.

@AnotherMatt92 What Arduino Board Support Package are you using?

Can not recreate this issue using a Pi Pico with the philhower core:
https://github.com/earlephilhower/arduino-pico
and running the `readMifare` example from the library:
```
Hello!
Found chip PN532
Firmware ver. 1.6
Waiting for an ISO14443A Card ...
Found an ISO14443A card
  UID Length: 4 bytes
  UID Value: 0x56 0x58 0x8F 0x7C

Seems to be a Mifare Classic card (4 byte UID)
Trying to authenticate block 4 with default KEYA value
Sector 1 (Blocks 4..7) has been authenticated
Reading Block 4:
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
```

AnotherMatt92, 2023-03-15:
Using the Philhower core 3.0.0

Any library example fails the same way (using iso1443a_uid.ino)
the only code change from the original example is commenting out the SPI nfc creation 
  Adafruit_PN532 nfc(PN532_SCK, PN532_MISO, PN532_MOSI, PN532_SS);
And uncommenting the I2C declaration
  Adafruit_PN532 nfc(PN532_IRQ, PN532_RESET);

Hello!
Didn't find PN53x board

the message about not finding the board takes 10+ seconds to reply too (no idea if that helps)

caternuson, 2023-03-15:
Using 3.1.0 release here for testing.

Try going through all other libraries and make sure they are up to date. Esp. the BusIO library:
https://github.com/adafruit/Adafruit_BusIO

AnotherMatt92, 2023-03-18:
Updated to 3.1.0, and BusIO is also upto date

Identical problem still exists

caternuson, 2023-03-20:
Must be some other difference, since nominally we are now running same hardware / example, but getting different results.

Is there anything else attached to the Pi Pico other than the PN532?

caternuson, 2023-07-31:
Closing due to lack of response. Could not recreate issue locally.

closed 2023-07-31 by caternuson