adafruit/Adafruit-PN532 #114: PN532 as_target change uid

lraton, 2023-04-14:
I'm using the  iso14443as_target example to emulate a card, but is possible to change the uid? because if i change  the ppse nothing change. The pn532 is connected via i2c

lraton, 2024-11-06:
No, i found that the pn change the first byte of the wild with a specific One for avoiding emulation. Probably for this is better using a pn7150

maxkarelov, 2024-11-06:
@lraton 

> i found that the pn change the first byte of the wild with a specific One for avoiding emulation

I saw that they use 0x08 for the first UID byte. but is it still possible to manage others? example from Seeed-PN532 doesn't work as expected but introduce managing left 3 bytes.

pn7150 dev board like this?

[image]

I am started recently to work with emulation and there is no much information, libraries and samples for getting started.
what can you recommend to make it more easy to get first results?

lraton, 2024-11-06:
I seem to have succeeded, but now I don't remember well that it's been a while since I've had my hands with it and I was considering changing to the 7150. The 7150 Is that from elechouse, it have the same size of 532 but Is purple.
[image]

I dont know very much about emulation, i've read some article from Salvador mendoza (https://salmg.net/) that Is the creator of the target function of this library