adafruit/Adafruit-PN532 #118: Compilation error when enabling debug in Adafruit_PN532 lib

jsiobj, 2023-07-09:
- Arduino board:  Adafruit M0 Proto

- Arduino IDE version (found in Arduino -> About Arduino menu): Using VSCode + PlatformIO

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):

When enabling debugging in Adafruit_PN532 lib uncommenting line 71/72
https://github.com/adafruit/Adafruit-PN532/blob/0bf348378b927fb6ce02079910826422d038405e/Adafruit_PN532.cpp#L71-L72

I get 2 compilation errors : 
`.pio\libdeps\adafruit_feather_m0\Adafruit PN532\Adafruit_PN532.cpp:957:28: error: invalid conversion from 'int8_t*' {aka 'signed char*'} to 'const byte*' {aka 'const unsigned char*'} [-fpermissive]`

For compilation to succeed, I forced conversion to (const uint8_t*) for _uid and _key when calling PrintHex here : 
https://github.com/adafruit/Adafruit-PN532/blob/0bf348378b927fb6ce02079910826422d038405e/Adafruit_PN532.cpp#L881-L888

With the change :
```
#ifdef MIFAREDEBUG
  PN532DEBUGPRINT.print(F("Trying to authenticate card "));
  Adafruit_PN532::PrintHex((const uint8_t*)_uid, _uidLen);
  PN532DEBUGPRINT.print(F("Using authentication KEY "));
  PN532DEBUGPRINT.print(keyNumber ? 'B' : 'A');
  PN532DEBUGPRINT.print(F(": "));
  Adafruit_PN532::PrintHex((const uint8_t*)_key, 6);
#endif
```

Not sure how to contribute / make a PR (code of conduct link in readme is broken) 
and not even sure this is the proper way to solve this issue anyway.

Regards
Jean-Sébastien

caternuson, 2023-08-01:
Also able to recreate this using Arduino IDE. Compiling the `readMifareClassic` library example for a target board of Feather M0:
```
Arduino/libraries/Adafruit-PN532/Adafruit_PN532.cpp:883:28: error: invalid conversion from 'int8_t*' {aka 'signed char*'} to 'const byte*' {aka 'const unsigned char*'} [-fpermissive]
  883 |   Adafruit_PN532::PrintHex(_uid, _uidLen);
      |                            ^~~~
      |                            |
      |                            int8_t* {aka signed char*}
```

caternuson, 2023-08-01:
Not sure `int8_t` is a good choice for these in the first place. Not seeing any reasons for them being signed.

@jsiobj Try removing the casts. Then change the declarations in `Adafruit_PN532.h` to:
```cpp
  uint8_t _uid[7];      // ISO14443A uid
  uint8_t _uidLen;      // uid len
  uint8_t _key[6];      // Mifare Classic key
```

That will most likely get past the compilation issue. But do other issue then arise? Like run time issues, etc?

jsiobj, 2023-09-10:
Hello @caternuson,

Indeed, it passes compilation successfully. I'll do some tests, hopefully next week and give the results.

I did not want to change variables types in the first place as just debugging code was not working and I was not sure of the implications. But I agree that changing the type should be the way to go.