adafruit/Adafruit-PN532 #119: Compilation error with RESET_PIN on GPIO0 on ESP32

danieljr-nh, 2023-07-20:
Custom board
Vs code + platformio
Adafruit_PN532 1.3.0

When I try to compile with RESET_PIN defined to GPIO_0, I get the error:
:call of overloaded 'Adafruit_PN532(int, int)' is ambiguous.

Regards
Daniel Jr.

caternuson, 2023-08-01:
Possibly something platformio specific?

There's no such overload in the library: 
```cpp
  Adafruit_PN532(uint8_t clk, uint8_t miso, uint8_t mosi,
                 uint8_t ss);                          // Software SPI
  Adafruit_PN532(uint8_t ss, SPIClass *theSPI = &SPI); // Hardware SPI
  Adafruit_PN532(uint8_t irq, uint8_t reset,
                 TwoWire *theWire = &Wire);              // Hardware I2C
  Adafruit_PN532(uint8_t reset, HardwareSerial *theSer); // Hardware UART
```
Unless platformio does something special for SPIClass and TwoWire.

caternuson, 2023-08-01:
@danieljr-nh Can you recreate this issue using the Arduino IDE?