adafruit/Adafruit-PN532 #121: IRQ functionality not working as expected.

mathaimon, 2023-11-14:
- Arduino board:  **Arduino Nano**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **2.2.1**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 
I'm using the  _readMifareClassicIrq_ example sketch from the library.
When the RFID tag is kept over the reader for longer than the **DELAY_BETWEEN_CARDS** duration, the reader does not seem to recognize the tag anymore and the whole setup has to be powered off and on to resume functionality.

[Here is the video of the issue](https://i.imgur.com/FPzsA6s.mp4)

caternuson, 2023-11-14:
I think what is happening is when the card is already present when `startListeningToNFC()` is called, it gets read immediately (by the PN532) and the IRQ asserted, etc. - but all that action is not handled by the example code. At that point, the example logic is sort of wrapped around the axle and stuck in a loop.

Try this modified version of the `startListeningToNFC()` function in the `readMifareClassicIrq` example:
```cpp
void startListeningToNFC() {
  // Reset our IRQ indicators
  irqPrev = irqCurr = HIGH;

  Serial.println("Starting passive read for an ISO14443A Card ...");
  if (!nfc.startPassiveTargetIDDetection(PN532_MIFARE_ISO14443A)) {
    Serial.println("No card found. Waiting...");
  } else {
    Serial.println("Card already present.");
    handleCardDetected();    
  }
}
```

mathaimon, 2023-11-16:
The above change seems to fix the issue.

caternuson, 2023-11-17:
OK, the example has been updated.

But hopefully it also shows the general behavior, so you won't run into a similar issue if you start adapting that example. The call to `startPassiveTargetIDDetection()` will actually also read any NFC that is also present. So be sure to check it's return and add logic and other calls as needed. That's really all that was done to fix the example.

mathaimon, 2023-11-20:
Can this issue be closed now

closed 2023-11-20 by caternuson