adafruit/Adafruit-PN532 #126: UART, I2C do not work with esp32

notsolowki, 2024-01-26:
I have only tried I2C and UART but I2C worked temporarily but now it won't work completely. It seems that with my pn532 card "the little square red one with multimode selection" is detected during boot. It reports the firmware version and pretends ready to read cards but it will not read anything. 

The weird part is it was reading them. If I go back to arduino it works perfect just not with the ESP32. What is the trick to making the reader work with esp32? 

I'm using official espressif esp32. Don't matter what mode I try, i2c and uart both detect firmware version but that's as far as it gets.. any ideas??

caternuson, 2024-01-26:
What sketch are you running and what output are you seeing in the Serial Monitor?

notsolowki, 2024-01-26:
I'm running the readmifare sketch, but I have tried other ones too. With the readMifare sketch the serial reports the boards firmware version and then waiting for card but never goes past that. 

With the format example it never goes past press any key to format. So strange it seemed like it was working for a awhile. I tried 4 other pn532 boards. All work on arduino but non work for esp32, at least  not any more. I tried many different esp32 of the model and same issue. Tried different pins etc. It always seems to report the board version but never rins any code in the loop

caternuson, 2024-01-26:
Just tested the `readMifare` sketch with a Feather ESP32 V2 connected via I2C and it worked as expected:
```
Hello!
Found chip PN532
Firmware ver. 1.6
Waiting for an ISO14443A Card ...
Found an ISO14443A card
  UID Length: 7 bytes
  UID Value: 0x04 0xFA 0x8C 0x42 0xD9 0x11 0x90

Seems to be a Mifare Ultralight tag (7 byte UID)
Reading page 4
01 03 A0 0C  ..⸮.
```

If it was working and then stopped working, then something must have changed. Please post this in the forums with photos of your setup:
https://forums.adafruit.com/

notsolowki, 2024-01-26:
Wow. I am so extremely sorry for asking you to waste your time. As it turns out I had another library conflicting with the adafruit library.... I just noticed it in my compiler log. 

It does indeed work as expected now. Again please forgive me for being ....... I greatly apologize and appreciate the time you took to help me. 

🤦‍♂️🤦‍♂️🤦‍♂️🤦‍♂️🤦‍♂️🤦‍♂️🤦‍♂️

closed 2024-01-26 by notsolowki

caternuson, 2024-01-26:
No worries. The PN532 is notoriously finicky, so can act weird and confusing sometimes. Glad it's working now.